insane-adding-machines/frosted #2: Frosted on STM32F411 Discovery board

DarkVegetableMatter, 2015-11-19:
I saw your message on the libopencm3 list and decided to check out Frosted. I got it running on my 411 Discovery board tonight and I had a couple of questions/issues

1) The PLL_CONFIG_VAL written to SYSREG_RCC_PLLCFGR is setting the clock too high for the 411. Which stm32f4xx are you testing with? Did you decide what to do about libopencm3 yet?

2) the objcopy option --pad-to=0x20000 in the top level makefile doesnt seem to work for me :( I expected the kernel. to be copied to a binary file and padded out to 128KB. Instead I get this:

$ arm-none-eabi-objcopy -O binary --pad-to=0x20000 kernel.elf kernel.bin
$ ls -al kernel.bin 
-rwxr-xr-x 1 mint mint 23756 Nov 19 01:44 kernel.bin

$ arm-none-eabi-objcopy -V
GNU objcopy (GNU Tools for ARM Embedded Processors) 2.24.0.20150304

So far I have the idling task running and 4 blinking LEDs. I want to try and get the Frost shell running next. Thanks for sharing, I think this is looking really interesting

danielinux, 2015-11-19:
Hello @DarkVegetableMatter - thanks for passing by. Let me quickly reply to these:
1) is one of the reason why we need a decent HAL. What I am doing now is hardcoding values to the hardware I have, which is totally wrong. I hope I did not cause you any HW damage :) 
My STM32 is the Discovery stm32f407.

I really really want to switch to libopencm3 as soon as possible, but the question "using" vs "deriving" libopencm3  is still a bit unclear to me. I hope the mailing list could shed some light on this so I can stop worrying about legal FUD and start integrating the library.

2) You are right, the padding is not working for stm32, it does work for qemu though. If you select stellaris as architecture, the same command seems to work. I might need more investigation on this. For now, I am flashing the two ELFs separately, which is what I suppose you did as well.

In order to run fresh, we need a better uart driver, and a correct pin muxing, which may depend on the board. I had no success so far running fresh on actual hardware.

You might have noticed you landed on a project at its very early stage, which is not yet usable, let alone useful. But I promise this is going to change soon. Stay tuned :)

DarkVegetableMatter, 2015-11-19:
Aha, --pad-to=0x8020000 is what we need.

"Pad the last section up to address <addr>" and our last section is at > 0x8000000

I understand Frosted is at an early stage. If I can help with testing on hardware etc etc please let me know

danielinux, 2015-11-19:
Yes! Thanks for spotting this. I will adapt the Makefile accordingly to cope with stm alignments. I'll ring a bell if/when we integrate opencm3. If this is going to be the case, there is no point in working around my crippled HAL any more.

danielinux, 2015-11-23:
closing due to migration to libopencm3

closed 2015-11-23 by danielinux