insane-adding-machines/frosted #4: Extending the config menu

DarkVegetableMatter, 2015-11-21:
-Changed the configuration menu to allow more flexibilitywhen selecting CPU & Platform
-Can now change size of Kernel partition (no longer fixed at 0x20000)
-RAM & Flash available to the App is now calculated (total - used by kernel)
-Hacked in a clock speed selection got STM32F4 (

Tested on STM32F411

Maybe you can take a look and tell me if this is going in the right direction? I still need to convert the Stellaris target but if this looks good then I can do that and some more testing on LPC & STM hardware

DarkVegetableMatter, 2015-11-21:
https://github.com/DarkVegetableMatter/frosted/tree/extendkconfig

danielinux, 2015-11-21:
Hello @DarkVegetableMatter - I started porting to libopencm3. It compiles and links but it's not quite working at the moment (missing clock initialization in kernel/frosted.c). I will be working on a lm3s port this weekend to ensure RCC can be configured via libopencm3 properly. STM32 port should be pretty complete though, so we can move ahead on that. Please could you have a look at this branch?
 https://github.com/insane-adding-machines/frosted/tree/libopencm3

You  can find me on IRC here: irc://irc.freenode.net/#frosted

danielinux, 2015-11-21:
cc: @maximevince

DarkVegetableMatter, 2015-11-21:
Hi

1) I needed to force libopencm3 to build with -mfloat-abi=soft. This builds
it from the cmd line

FP_FLAGS="-mfloat-abi=soft" make -C kernel/libopencm3

Please consider removing libopencm3 from the clean make target (maybe a
cleanall or cleanlibopencm3 ?)

2) I changed Reset_Handler to reset_handler in def_irq.py - do we want to
use libopencm3 startup code or yours?

https://github.com/DarkVegetableMatter/frosted/tree/dvm_libopencm3

I will see if I can get this build to boot,  hope I'm doing the right thing

danielinux, 2015-11-22:
Hi,
I am a few steps ahead in the branch that your dvm_libopencm3 branch does not seem to take into account! Please ensure you are based on the branch:
https://github.com/insane-adding-machines/frosted/tree/libopencm3 

As for question (1): I had noticed that cleaning the lib all the time slows down the flow. The branch already has a "libclean" separate target. Perhaps the best way to force FP_FLAGS is from the main Makefile while doing make -C kernel/libopencm3

(2) Startcode should be the one from openocm3. There is no kernel/hal directory anymore. The linker scripts are now in kernel/ld. I just committed a new linker script for stm32, but there is more work to do for startup code and uart.

Also, please note that the libopencm3 submodule points to this repository:
https://github.com/insane-adding-machines/libopencm3/tree/qemu

Where I have extended the lib to add support for LM3S clock settings and UART. The latest status with this combo is I can have the full synthetic port to work.

The plan is to propose a pull request to libopencm3 official when the port is stable.
The UART interface for all the ports e are going to support needs to be extended to support  rx_ready and a irq_clear commands, as it already happens now in my branch for qemu/lm3s.

DarkVegetableMatter, 2015-11-22:
Ok, with a bit more tweaking I am now using the libopencm3 startup code and
linker script my code starts and calls frosted main() :)

DarkVegetableMatter, 2015-11-22:
hmm, seems I cannot send on that irc channel... dont know why

danielinux, 2015-11-22:
query

closed 2015-11-22 by danielinux

maximevince, 2015-11-23:
:+1: awesome

DarkVegetableMatter, 2015-11-23:
Hi

STM32F4 libopencm3 GPIO is now working on the HW. It's a bit basic & we
need to decide how configuration (AF, pullup/down, open collector etc) will
be passed through the ioctl argument. Macros to pack bits into a 32 byte
number? Pointer to a structure?

LPC17xx libopencm3 GPIO is also working on the HW. Same comments as STM32F4
above

LPC17xx clock support in libopencm3 is basically 0 so I have brewed up some
code that, I believe, is clocking the CPU at 100MHz. There is also a typo
in the LPC17xx clock.h file that means it cant be used to compile as-is

I haven't committed this code because I haven't figured out qemu yet - it
starts and a black window appears but I haven't figured out how to connect
a terminal to it yet :S

I think we will need a 'platform init' function to group the startup code
for different machines. For example, the mbed/seeedpro have the LEDs
connected to the same pins but the XPresso (LPC1769) only has one LED and
it's connected to a different pin. Happily the config code now allows the
user to select a board (STM32F4Discovery, seedpro etc etc) so we could use
that

maximevince, 2015-11-23:
Hi @DarkVegetableMatter,

You can try 'make qemu' or 'make qemu2' as defined in the Makefile.
They should spawn you with a qemu UART terminal.

```
qemu: image.bin 
	qemu-system-arm -semihosting -M lm3s6965evb --kernel image.bin -serial stdio -S -gdb tcp::3333

qemu2: image.bin
	qemu-system-arm -semihosting -M lm3s6965evb --kernel image.bin -serial stdio
```

danielinux, 2015-11-23:
This is fixed - the code is committed :) In the documentation, we will  need to specify that minimum qemu version is 2.2.x