insane-adding-machines/frosted #8: Adding '.' link in new directories

Maldus512, 2015-12-14:
I'm a graduating computer science student and I'd like to collaborate to frosted for an internship. To get started I'm trying to add '.' and '..' links to directories.
I simply added the fno_link call to create '.' inside fno_create_dir; everything seems to work fine for the directories created directly by fresh() (/home and /home/test) as ls displays the '.' link to self.
However, creating a directory from the shell with the mkdir command does not work, and I'm not able to figure out why.
From what I understood by debugging the new directory is created correctly but the link is not, and somehow the process of creating the link ends up corrupting the fname field in the new directory fnode ( random characters ).
I think I narrowed the problem to one of the two fno_search calls (sometimes it's the former, sometimes the latter ), the first in fno_link and the second in fno_create_file: when the new directory fnode ( which is to be linked as '.' ) is found - in the last recursive call to _fno_search - its fname field changes to random characters ( just that field ). Because of that the link creation fails, as the original name to be linked cannot be found any more.
The directory with the random name exists and it is accessible through cd.

Note that this is my first time debugging with gdb or even handling some serious kernel code, so I could easily be wrong in my deductions.

danielinux, 2015-12-16:
Hello, your links created at startup were being overwritten by mount.
I added a new function `mkdir_links` containing your code in `vfs.c` which is being called upon mount and mkdir, and seems to work as expected.

Feel free to complete with parent's link

danielinux, 2015-12-16:
lol for "serious kernel code" :+1:

closed 2015-12-16 by danielinux