insane-adding-machines/frosted #22: Add support for route.h in libC

danielinux, 2016-03-19:
- [ ] Add support for route.h in toolchain
 - [ ] write userland netutils/route to add/rm/list routes

danielinux, 2016-05-11:
Support for net/route.h in newlib added:
https://github.com/insane-adding-machines/newlib/commit/18862e0fd1b8531f269f72b700195a6d65bdb99c

danielinux, 2016-05-11:
route binary: work in progress
https://github.com/insane-adding-machines/frosted-userland/blob/master/netutils/route.c

currently missing parameters parsing from command line.

danielinux, 2016-05-11:
@NicoLiberato - were you going to work on this?

NicoLiberato, 2016-05-15:
As soon I have a fast Internet connection. I will provide you asap.

danielinux, 2016-06-10:
@NicoLiberato ping?

NicoLiberato, 2016-06-10:
Hi Danielinux, I have high bandwidth, let's back on track

danielinux, 2016-08-01:
Continuing on #62

closed 2016-08-01 by danielinux