insane-adding-machines/frosted #37: Implement dlclose(), dlerror(), dlopen(), and dlsym()

danielinux, 2016-05-19:
Straight from the wishlist. 
http://linux.die.net/man/3/dlopen

@maximevince please provide some starting pointers from your deep knowledge of the obscurities (elf2flt, bflt, ...)!

sferrini, 2016-10-06:
Hey @maximevince are you working on this?

Unfortunately I've no in-depth experience in dynamic loading but if you guide me, I can help you implementing this.
What file format have you choose for dynamic libraries?

danielinux, 2016-10-06:
The format we use for execution in place is bFLT. For the libraries, it'd be nice if we could execute code in place from flash as well, so the objects can be placed somewhat in xipfs, preserving the symbol addresses. LibC should act accordingly, and provide those addresses when the symbol is requested via dlsym(). 

I think that a harder part is making (part of) libc dynamic, as we need to identify the symbols used by all the apps in the Xipfs, and discard the rest; which obviously can't be done automatically by the linker at this time.