insane-adding-machines/frosted #49: [malloc] corrupted malloc_block list

danielinux, 2016-06-26:
possibly due to concurrency, the malloc_block[] list array may end up containing invalid entries in the tail (wrong last ->next pointer?)

danielinux, 2016-06-26:
@maximevince cannot find the root cause. Seems to happen on both kernel and user pool. Somewhat an internal operation is corrupting the "->next" pointer for the last block.
Workaround for now is checking if blk->magic is valid while visiting the list.

brabo, 2016-06-26:
Workaround is not working on my end. (F429i-disco)

danielinux, 2016-06-27:
@maximevince is investigating on the issue. Perhaps we can change the malloc design a bit (e.g. do we need f_compact after every free?). Most likely there is a critical section we are not taking into account in one of the operations (free, split, merge, compact).

danielinux, 2016-07-01:
Fixed in eed4d79

closed 2016-07-01 by danielinux