insane-adding-machines/frosted #50: RNG: Rewrite RNG driver.

brabo, 2016-06-26:
https://github.com/insane-adding-machines/libopencm3/commit/6c0fa767776174e6e6fa54a1c7a24acdfcfb2055 introduces clean functions to (en|dis)able the RNG and to read from it, allowing the driver code to be agnostic about registers and be more clean.

brabo, 2016-07-06:
Expanded unicore-mx rng support to use convenience functions.
Refactored rng driver and added f2 support to frosted.
Need to handle errors correctly still.

danielinux, 2016-07-07:
What is still missing in the current implementation: 
- [ ] Activate interrupt before the calling process suspends upon BUSY/underseed states
- [ ] Implement ISR to trigger task_resume() and disable IRQ again
- [ ] Implement a poll function that returns 1 if rng can be read (just check the flags..) or 0 if busy/underseed

If you want to implement #55 perhaps it's worth keeping both mechanism in place, so /dev/random can be attached to either the RNG directly or to libfortuna, depending whether CONFIG_LIBFORTUNA is enabled.

danielinux, 2016-07-16:
@brabo : fixed via fortuna branch.

closing.

closed 2016-07-16 by danielinux