insane-adding-machines/frosted #52: fatal error: stm32f4_usbdef.h

FabHer74, 2016-07-04:
Hello,

I'm new to "frosted" and I would like to use it with a STM32 µC (in a first time on a Discovery board and after a Nucleo).

Unfortunatly, I have followed the "getting started" steps and I had the issue hereafter, it seems that a file is not there (stm32f4_usbdef.h).

I'm not able to find it in the frosted and arm-frosted-eabi.

Any chance to find it somewhere else ?
I tried to use the stm32f4_usb.h but it doesn't help.

Thank for your help.

_fabien@fanher:~/frosted$ make
arm-frosted-eabi-gcc -DFLASH_ORIGIN=0x08000000 -DSTM32F4 -mcpu=cortex-m4 -mfloat-abi=soft -DCONFIG_RNG -DFLASH_ORIGIN=0x08000000 -DAPPS_ORIGIN=$(( 64 * 1024)) -DCONFIG_KRAM_SIZE=64 -DCONFIG_SYS_CLOCK=84000000 -mthumb -mlittle-endian -mthumb-interwork -DCORE_M3 -fno-builtin -ffreestanding -DSYS_CLOCK=84000000 -DCORTEX_M3 -DFROSTED -Ikernel/unicore-mx/include -Ikernel -Iinclude -I. -mthumb -mlittle-endian -mthumb-interwork -DCORE_M3 -DBOARD_stm32f4 -DSTM32F4 -DCONFIG_KMEM_SIZE= -DCONFIG_TASK_STACK_SIZE=1024 -DCONFIG_KLOG_SIZE=128 -DCONFIG_KLOG -DCONFIG_MEMFS -DCONFIG_SYSFS -DCONFIG_DEVNULL -DCONFIG_DEVSTM32F4I2C -DCONFIG_DEVUART -DCONFIG_DEVGPIO -DCONFIG_DEVSTM32F4DMA -DCONFIG_RNG -Ikernel -Iinclude -I. -Ikernel/unicore-mx/include/unicore-mx -Ikernel/unicore-mx/include -fno-builtin -ffreestanding -nostdlib -ggdb -ggdb3 -Ikernel/frosted-headers/include -DCONFIG_KLOG -DCONFIG_MEMFS -DCONFIG_SYSFS -DCONFIG_DEVNULL -DCONFIG_DEVSTM32F4I2C -DCONFIG_DEVUART -DCONFIG_DEVGPIO -DCONFIG_DEVSTM32F4DMA -DCONFIG_RNG   -c -o kernel/stm32f4/stm32f4x1discovery.o kernel/stm32f4/stm32f4x1discovery.c
kernel/stm32f4/stm32f4x1discovery.c:358:28: fatal error: stm32f4_usbdef.h: No such file or directory
compilation terminated.
make: *** [kernel/stm32f4/stm32f4x1discovery.o] Erreur 1__

danielinux, 2016-07-05:
Hi @FabHer74 

Sorry, but the person who was maintaining the port for the F4x1 Discovery has left the team, and the port has not been checked lately. 

That include statement in the porting file is no longer needed, I am committing a patch for this.

danielinux, 2016-07-05:
@FabHer74 please try compiling the latest master.

closed 2016-07-05 by danielinux

FabHer74, 2016-07-05:
Fine, it compiles.

Previously, I already comment that line to go further but it's better to 
get your confirmition and patch.

Thanks

Fab

commit c9ab76f8 (2016-07-05): [STM32F4x1] Removed include statement of old usb definition header.

Fixes #52