insane-adding-machines/frosted #53: fatal error: sys/reboot.h

FabHer74, 2016-07-04:
Hello,

The file reboot.h is missing when I compile my project (result hereafter).
My git clone is up-to-date (git submodule init && git submodule update).

I use arm-frosted-eabi-5.3.0_16.06.004 but the file reboot.h is not included.

Any idea ?

Thanks

_arm-frosted-eabi-gcc -mthumb -mlittle-endian -mthumb-interwork -ffunction-sections -mcpu=cortex-m3 -DCORE_M3 -D__frosted__ -nostartfiles -fPIC -mlong-calls -fno-common -msingle-pic-base -mno-pic-data-is-text-relative -Wstack-usage=1024 -ggdb -I../lib/include   -c -o reboot.o reboot.c
reboot.c:23:24: fatal error: sys/reboot.h: No such file or directory
compilation terminated.
make[2]: *** [reboot.o] Erreur 1_

danielinux, 2016-07-05:
Hi again @FabHer74 

16.06.004 is not the latest toolchain. The reboot() syscall has been added in version 16.07, which you can find here as usual:
https://github.com/insane-adding-machines/crosstool-ng/releases
(Check toolchain changelog, a few new features have been added in our July release).

If you want to use the newest syscalls, you need a new toolchain. The alternative would be to run menuconfig from the `frosted-userland` directory and exclude the reboot and the renice binaries from your XIPFS build.

HTH.

closed 2016-07-05 by danielinux

FabHer74, 2016-07-05:
Hello,

It compiles now, thanks.
I didn't saw the toolchain 2 days ago.

Fab

danielinux, 2016-07-05:
yes, we move rather fast :)