insane-adding-machines/frosted #74: Find a way to reduce # of fnodes when mounting filesystems

danielinux, 2016-08-06:
When a filesystem is mounted, frosted scans the whol content and creates fnodes accordingly.

Perhaps the fnodes should only be allocated during lookup (fno_search), or when opendir/readdir is called, in order to decrease the number of active fnodes, and the memory required.

ichigo663, 2017-08-26:
@danielinux  @brabo  what do you think? the problem here is obvious... any feedback on how to get around this? what does readdir_r() call behind the curtain?

danielinux, 2017-08-26:
https://github.com/insane-adding-machines/frosted/blob/master/kernel/vfs.c#L641

brabo, 2017-08-27:
and opendir does call fno_search, so before readdir() the loading has happened unless I am missing something.

ichigo663, 2017-08-30:
opendir() just does fno_search on the directory file, it doesn't look for the files inside the directory, that's why it doesn't work just like this.

brabo, 2017-08-30:
if lookup is called with a directory it would populate that on FS level. i am not sure i see any problem.
the thing i wonder is though, if opendir is called on say '/mnt/sub1/sub2/sub3/sub4', what exactly do we want/need to cache from lookup? that whole tree and their contents?