insane-adding-machines/frosted #81: LTDC: buffer allocation & blank

sferrini, 2016-09-28:
Reading the `stm32f7_ltdc` driver implementation I was trying to understand the `screen_buffer` and how this driver in general works.

And I've noticed at this [line](https://github.com/insane-adding-machines/frosted/blob/master/kernel/drivers/stm32f7_ltdc.c#L59) that the buffer allocation is done calculating the size like so: `fb->var.xres * fb->var.xres * (fb->var.bits_per_pixel/8)`. Shouldn't be calculated also involving the `yres`? So like `fb->var.xres * fb->var.yres * (fb->var.bits_per_pixel/8)`?

Also in the function `ltdc_blank` at this [line](https://github.com/insane-adding-machines/frosted/blob/master/kernel/drivers/stm32f7_ltdc.c#L169) shouldn't `fb->var.bits_per_pixel` be `(fb->var.bits_per_pixel/8)`?

Maybe I get the implementation in a wrong way, so sorry in advance if so.

@maximevince can you please have a look?

Let me know what you think, and I'll do a PR if needed.

Thanks

danielinux, 2016-09-29:
@sferrini thanks for reporting. Indeed let's wait for @maximevince comment.

maximevince, 2016-09-29:
@sferrini hey, thanks for reviewing our code, we like that.
that is a huge typo indeed, nicely spotted!
if you make another pull request, we'll integrate that.

maximevince, 2016-09-29:
Both of your suggestions are correct, by the way!

The LTDC implementation right now is _very_ basic; i just rushed into it, to get DOOM working :)
So any improvements are welcome!

sferrini, 2016-09-29:
@maximevince It's a pleasure! I'll do the PR in a bit 😉

brabo, 2016-09-29:
:D

PR #82 (2016-09-29): Fixed LTDC driver buffer allocation & blank
This PR fixes #81
closed 2016-09-29 by maximevince