insane-adding-machines/frosted #84: Ethernet doesn't seem to work on STM32f746 nucleo/disco

k0d, 2016-10-18:
Have tried just about every setting but no traffic seems to work. Have you had it working at any point on these boards?

danielinux, 2016-10-18:
Hello, I noticed that too while testing last week. This is confirmed.

The same driver is working on:
- STM32F407-discovery
- STM32F769-discovery
- STM32F746-nucleo

So I suppose there are just some setting missing/misplaced.

k0d, 2016-10-18:
It wasn't working on STM32F746-nucleo for me, I'll try again now and come back with more info.

k0d, 2016-10-18:
It still doesn't work on STM32F746-nucleo: here is my .config file Do you see anything wrong?

#
# Automatically generated file; DO NOT EDIT.
# FROSTED Kernel Configuration
#

#
# Compiler options
#
GDB_CFLAG=y
OPTIMIZE_NONE=y
# OPTIMIZE_SIZE is not set
# OPTIMIZE_PERF is not set

#
# Platform Selection
#
# ARCH_LM3S is not set
# ARCH_LPC17XX is not set
# ARCH_STM32F4 is not set
ARCH_STM32F7=y
ARCH_STM32F746_NG=y
# ARCH_STM32F769_NI is not set
FLASH_SIZE_1MB=y
RAM_SIZE_320KB=y
ARCH_STM32F746=y
CLK_216MHZ=y
# MACH_STM32F746Discovery is not set
# MACH_STM32F769Discovery is not set
MACH_STM32F746Nucleo144=y
# STM32F7_SDRAM is not set

#
# Kernel Configuration
#
KFLASHMEM_SIZE=128
KRAMMEM_SIZE=64
# TASK_STACK_SIZE_1K is not set
TASK_STACK_SIZE_2K=y
# TASK_STACK_SIZE_4K is not set
# TASK_STACK_SIZE_8K is not set
MPU=y

#
# IPC features
#
SIGNALS=y
PIPE=y
# SOCK_UNIX is not set

#
# Debugging options
#
KLOG=y
KLOG_SIZE=256
MEMFAULT_DBG=y
HARDFAULT_DBG=y
# STRACE is not set

#
# Filesystems
#
SYSFS=y
MEMFS=y
XIPFS=y
FATFS=y
FAT32=y
FAT16=y

#
# Networking
#
SOCK_INET=y
# TCPIP_MEMPOOL_YN is not set
PICOTCP=y

#
# picoTCP configuration
#
CONFIG_PICOTCP_IPV4=y
# CONFIG_PICOTCP_IPV6 is not set
CONFIG_PICOTCP_TCP=y
CONFIG_PICOTCP_UDP=y
CONFIG_PICOTCP_DNS=y
# CONFIG_PICOTCP_MCAST is not set
# CONFIG_PICOTCP_NAT is not set
# CONFIG_PICOTCP_IPFILTER is not set
CONFIG_PICOTCP_LOOP=y
# CONFIG_PICOTCP_DEBUG is not set

#
# Device Drivers
#
DEVNULL=y
DEVUART=y
# USART_1 is not set
# USART_2 is not set
# USART_3 is not set
USART_6=y
# DEVSTM32USB is not set
DEVF7ETH=y
ETH_DEFAULT_IP="10.0.1.91"
ETH_DEFAULT_NM="255.255.255.0"
ETH_DEFAULT_GW="10.0.1.1"
DEVSTM32SDIO=y
# DEVSTM32I2C is not set
# DEV_RANDOM is not set
# DEVFRAMEBUFFER is not set

#
# Power Management
#

#
# Power Management requires CPU Timer support
#

danielinux, 2016-10-18:
I will test as soon as possible and let you know

k0d, 2016-10-19:
Fixed for both, but ping stops working sometimes on nucleo, will open new issue for that

closed 2016-10-19 by k0d