insane-adding-machines/frosted #94: Implement pthreads

danielinux, 2016-10-21:
https://linux.die.net/man/7/pthreads

danielinux, 2016-10-23:
TODO:
- [ ] implement pthread_join properly
- [ ] multiplex signals properly

danielinux, 2016-10-28:
@ichigo663 - since you are busy with the scheduler, is it perhaps interesting for you to continue on this one?
We need proper join/detach implementation. One thread in each group can call join, maybe you can reuse the ZOMBIE state if the thread is detachable.

ichigo663, 2016-10-28:
Ok! I'll start looking into it!

danielinux, 2017-01-04:
Looks like only pthread_kill is urgently missing (along with proper signal distribution).

danielinux, 2017-03-14:
This can be closed for now, as micropython successfully compiled with thread support. 

More pthread-related feature requests belong to new issues.

closed 2017-03-14 by danielinux