insane-adding-machines/frosted #107: [Wishlist] port µGFX

danielinux, 2016-11-17:
"µGFX is a library to interface different displays and touchscreens to your embedded application"

http://ugfx.io
https://github.com/Tecnologic/uGFX

danielinux, 2016-11-17:
Hello @k0d - could you update us on the status of the port? Do you need extra helping hands?

sferrini, 2016-11-17:
Hey @danielinux, how will the µGFX library be linked to the kernel? It will be just an executable? Because I read the [license](http://ugfx.io/license.html) and for commercial use it's a [paid](http://ugfx.io/pricing) one.

sferrini, 2016-11-17:
Also I guess the upstream repo is this one: https://git.ugfx.io/uGFX/uGFX

danielinux, 2016-11-17:
@sferrini no link to the kernel... this is a userspace project, so no worries about license issues.

sferrini, 2016-11-17:
Oh right @danielinux! I didn't see it, thanks 👍

k0d, 2016-11-18:
No update as I've returned back to work after sickleave and there is 20x the normal workload right now. So Frosted dev is on hold for a short time for me.

danielinux, 2016-11-19:
@k0d ugh... sorry to hear. Hope you are feeling better now.

danielinux, 2017-01-20:
Currently being handled by the uGFX team, so closing this for now.

closed 2017-01-20 by danielinux