insane-adding-machines/frosted #113: Can't `make menuconfig` on macOS Darwin Kernel Version 16.3.0

grapherd, 2017-01-05:
Still trying to fix this problem.

```
➜  frosted git:(master) make menuconfig
././lxdialog/check-lxdialog.sh -check gcc -DCURSES_LOC="<ncurses.h>" -DLOCALE -DKBUILD_NO_NLS -I. -DCONFIG_=\"\"
gcc -DCURSES_LOC="<ncurses.h>" -DLOCALE -DKBUILD_NO_NLS -I. -DCONFIG_=\"\"   -c lxdialog/checklist.c -o lxdialog/checklist.o
gcc -DCURSES_LOC="<ncurses.h>" -DLOCALE -DKBUILD_NO_NLS -I. -DCONFIG_=\"\"   -c lxdialog/inputbox.c -o lxdialog/inputbox.o
gcc -DCURSES_LOC="<ncurses.h>" -DLOCALE -DKBUILD_NO_NLS -I. -DCONFIG_=\"\"   -c lxdialog/menubox.c -o lxdialog/menubox.o
gcc -DCURSES_LOC="<ncurses.h>" -DLOCALE -DKBUILD_NO_NLS -I. -DCONFIG_=\"\"   -c lxdialog/textbox.c -o lxdialog/textbox.o
gcc -DCURSES_LOC="<ncurses.h>" -DLOCALE -DKBUILD_NO_NLS -I. -DCONFIG_=\"\"   -c lxdialog/util.c -o lxdialog/util.o
gcc -DCURSES_LOC="<ncurses.h>" -DLOCALE -DKBUILD_NO_NLS -I. -DCONFIG_=\"\"   -c lxdialog/yesno.c -o lxdialog/yesno.o
gcc -DCURSES_LOC="<ncurses.h>" -DLOCALE -DKBUILD_NO_NLS -I. -DCONFIG_=\"\"   ./mconf.o ./zconf.tab.o ./lxdialog/checklist.o ./lxdialog/util.o ./lxdialog/inputbox.o ./lxdialog/textbox.o ./lxdialog/yesno.o ./lxdialog/menubox.o  -o ./mconf
Undefined symbols for architecture x86_64:
  "_acs_map", referenced from:
      _dialog_checklist in checklist.o
      _print_arrows in checklist.o
      _dialog_clear in util.o
      _draw_box in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _dialog_yesno in yesno.o
      ...
  "_cbreak", referenced from:
      _init_dialog in util.o
  "_delwin", referenced from:
      _dialog_checklist in checklist.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _dialog_yesno in yesno.o
      _dialog_menu in menubox.o
  "_doupdate", referenced from:
      _dialog_checklist in checklist.o
  "_endwin", referenced from:
      _init_dialog in util.o
      _end_dialog in util.o
  "_flash", referenced from:
      _dialog_inputbox in inputbox.o
  "_getcurx", referenced from:
      _init_dialog in util.o
      _print_autowrap in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _print_arrows in menubox.o
  "_getcury", referenced from:
      _init_dialog in util.o
      _print_autowrap in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _print_arrows in menubox.o
  "_getmaxx", referenced from:
      _show_help in mconf.o
      _dialog_checklist in checklist.o
      _dialog_clear in util.o
      _init_dialog in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _print_position in textbox.o
      ...
  "_getmaxy", referenced from:
      _dialog_checklist in checklist.o
      _attr_clear in util.o
      _dialog_clear in util.o
      _init_dialog in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _print_position in textbox.o
      ...
  "_has_colors", referenced from:
      _color_setup in util.o
      _draw_shadow in util.o
  "_init_pair", referenced from:
      _init_one_color in util.o
  "_initscr", referenced from:
      _init_dialog in util.o
  "_keypad", referenced from:
      _dialog_checklist in checklist.o
      _init_dialog in util.o
      _on_key_esc in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _dialog_yesno in yesno.o
      _dialog_menu in menubox.o
      ...
  "_newwin", referenced from:
      _dialog_checklist in checklist.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _dialog_yesno in yesno.o
      _dialog_menu in menubox.o
  "_nodelay", referenced from:
      _on_key_esc in util.o
  "_noecho", referenced from:
      _init_dialog in util.o
  "_scrollok", referenced from:
      _dialog_checklist in checklist.o
      _dialog_menu in menubox.o
      _do_scroll in menubox.o
  "_start_color", referenced from:
      _color_setup in util.o
  "_stdscr", referenced from:
      _show_help in mconf.o
      _dialog_checklist in checklist.o
      _dialog_clear in util.o
      _init_dialog in util.o
      _end_dialog in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      ...
  "_subwin", referenced from:
      _dialog_checklist in checklist.o
      _dialog_textbox in textbox.o
      _dialog_menu in menubox.o
  "_ungetch", referenced from:
      _on_key_esc in util.o
  "_waddch", referenced from:
      _dialog_checklist in checklist.o
      _print_item in checklist.o
      _print_arrows in checklist.o
      _attr_clear in util.o
      _dialog_clear in util.o
      _print_title in util.o
      _print_button in util.o
      ...
  "_waddnstr", referenced from:
      _print_item in checklist.o
      _print_arrows in checklist.o
      _dialog_clear in util.o
      _print_title in util.o
      _print_autowrap in util.o
      _print_button in util.o
      _dialog_inputbox in inputbox.o
      ...
  "_wattrset", referenced from:
      _dialog_checklist in checklist.o
      _print_item in checklist.o
      _print_arrows in checklist.o
      _attr_clear in util.o
      _dialog_clear in util.o
      _print_title in util.o
      _print_button in util.o
      ...
  "_wbkgdset", referenced from:
      _dialog_textbox in textbox.o
      _print_position in textbox.o
      _dialog_menu in menubox.o
  "_wclrtoeol", referenced from:
      _print_line in textbox.o
      _do_print_item in menubox.o
  "_wgetch", referenced from:
      _dialog_checklist in checklist.o
      _on_key_esc in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _dialog_yesno in yesno.o
      _dialog_menu in menubox.o
  "_winch", referenced from:
      _draw_shadow in util.o
  "_wmove", referenced from:
      _dialog_checklist in checklist.o
      _print_item in checklist.o
      _print_arrows in checklist.o
      _print_buttons in checklist.o
      _attr_clear in util.o
      _dialog_clear in util.o
      _end_dialog in util.o
      ...
  "_wnoutrefresh", referenced from:
      _dialog_checklist in checklist.o
      _dialog_clear in util.o
      _draw_shadow in util.o
      _dialog_textbox in textbox.o
      _print_page in textbox.o
      _dialog_menu in menubox.o
  "_wprintw", referenced from:
      _print_item in checklist.o
      _print_position in textbox.o
  "_wrefresh", referenced from:
      _dialog_checklist in checklist.o
      _print_item in checklist.o
      _print_buttons in checklist.o
      _end_dialog in util.o
      _dialog_inputbox in inputbox.o
      _print_buttons in inputbox.o
      _refresh_text_box in textbox.o
      ...
  "_wscrl", referenced from:
      _dialog_checklist in checklist.o
      _do_scroll in menubox.o
  "_wtouchln", referenced from:
      _attr_clear in util.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [mconf] Error 1
make: *** [menuconfig] Error 2
```

testing method from: https://github.com/espressif/esp-idf/issues/25

danielinux, 2017-01-05:
Hi, thanks for reporting this. I am not able to help myself, as my macbook runs Linux only. I will ask around to see if someone can reproduce this.

g5pw, 2017-01-11:
@grapherd It looks like you may not have `ncurses` installed or `pkg-config` is not finding it. What is the output of the following commands on your machine?

```bash
pkg-config --libs ncurses
pkg-config --libs ncursesw
```

danielinux, 2017-01-20:
@grapherd Any luck with this? Did you try g5pw solution?
Thanks,

grapherd, 2017-01-20:
@danielinux sorry for late reply, my mac was broke and sent to repair......

I think the problem isn't @g5pw says, I've testing by applying this [patch](https://github.com/f9micro/f9-kernel/pull/119), it have similar problem, but after trying pkg-config, it still pop out the same question.

I'm not sure is makefile problem or not, I get a try when my mac comeback.

also, will you attend FOSDEM2017?

Thanks,

danielinux, 2017-01-20:
Yes, I'll be there, I am part of the staff. Also relevant: @brabo 's talk in the IoT devroom https://fosdem.org/2017/schedule/event/iot_frosted/

danielinux, 2017-03-14:
@grapherd  any news on this? Can we close the issue? We don't have any Darwin, so we can't really test...

danielinux, 2017-03-30:
Not enough information. Closing.

closed 2017-03-30 by danielinux