insane-adding-machines/frosted #114: Can't compile with arm-frosted-eabi-gcc for a cortex M7

nathanLoretan, 2017-01-06:
Hello, perhaps it's my bad but always when I try to compile for a cortex M7, I want to use a board stm32f746-discovery, I have this error:
 Assembler messages:
 Error: unknown cpu `cortex-m7'
 Error: unrecognized option -mcpu=cortex-m7
 make[1]: *** [init.o] Error 1
 make: *** [binutils] Error 2

When I compile for cortex M4 or M3 I have no problem it's only for cortex M7.

danielinux, 2017-01-06:
Currently we don't have multi-arch support for our compiler. You can safely compile for your M7 target using cortex-m3 flags. @maximevince  is working on a multi-arch setup for the compiler.

Can you describe what you are trying to do? Are you compiling an application or a library for frosted? Any specific reason why you'd need M7 optimizations?

danielinux, 2017-01-06:
FYI: This is the suggested set of flags to compile userspace apps and libraries for frosted:

CFLAGS:
`-mthumb -mlittle-endian -mthumb-interwork -ffunction-sections -mcpu=cortex-m3 -DCORE_M3 -D__frosted__ -nostartfiles  -fPIC -mlong-calls -fno-common -msingle-pic-base -mno-pic-data-is-text-relative -Wstack-usage=1024`

LDFLAGS:
`-fPIC -mlong-calls -fno-common -Wl,-elf2flt -lgloss`

nathanLoretan, 2017-01-06:
I try to make a port for using ugfx library and frosted. to do this, I need to compile the ugfx library with specifying cortex-m7. And for compile the library I use arm-frosted-eabi-gcc. It will work if I compile the library ugfx with arm-none-eabi-gcc and the kernel frosted with arm-frosted-eabi-gcc?

danielinux, 2017-01-06:
Any specific reason to compile with -mcpu=cortex-m7? Why can't you compile with -m3?

nathanLoretan, 2017-01-06:
I will try to compile with -m3. cortex-m7 is the automatically specifying when you choose the board stm32f746-discovery on the library ugfx.

danielinux, 2017-01-06:
@nathanLoretan be careful, because you are not compiling a standalone, baremetal application, but rather a unix userspace app, which should be by definition generic (except maybe for the resolution of the screen). 

You should not need anything that is board specific, all you need is accessing /dev/fb0 which is a generic (memory mapped) device, using standard system calls (open/lseek/write).

Please have a look on how we are doing it in Id software's DOOM port:
https://github.com/insane-adding-machines/DOOM/blob/master/frosted-doom/i_video_fbdev.c#L363

closed 2017-01-09 by danielinux