insane-adding-machines/frosted #116: undefined reference to 'clock_gettime'

nathanLoretan, 2017-01-26:
Hello, I try to use the function clock_gettime(). I define _POSIX_TIMERS and I include <time.h> but I always have the error: undefined reference to 'clock_gettime'.

danielinux, 2017-01-27:
Hi! In fact clock_gettime/clock_settime/clock_getres are not yet implemented in libC.

I will turn this issue in a feature request and take care of it ASAP. Meanwhile, any chance you can use gettimeofday() ?

Thanks

d

danielinux, 2017-03-05:
Not fixed yet for us, so better not close it @nathanLoretan

brabo, 2017-03-07:
ok, this is next up on my frosted-calendar!

brabo, 2017-03-07:
@nathanLoretan what resolution are you guys expecting clock_gettime() to have? assuming XSI-conformance or?

nathanLoretan, 2017-03-07:
CLOCK_REALTIME

brabo, 2017-03-07:
@nathanLoretan that is a clock, not the resolution. i ask because of:
"Although the value of CLOCKS_PER_SEC is required to be 1 million on all XSI-conformant systems, it may be variable on other systems, and it should not be assumed that CLOCKS_PER_SEC is a compile-time constant"
if you assume 1 million per second, it may cause issues as our system clock resolution is only 1000/sec, i.e. millisecond resolution, and i noticed there is no clock_getres() call in uGFX to determine what resolution the system has.

danielinux, 2017-03-07:
@brabo we should stick with 1ms resolution for these calls anyhow

brabo, 2017-03-08:
@nathanLoretan done, closing this issue ;)

brabo, 2017-03-08:
@nathanLoretan note, just include sys/time.h for now, not time.h as posix specifies.

danielinux, 2017-03-14:
Related to #120

brabo, 2017-03-14:
time.h is there now.

closed 2017-03-14 by danielinux