insane-adding-machines/frosted #117: nanosleep() always returns -1

nathanLoretan, 2017-03-05:
Hello, the function nanosleep() always returns -1. To try, I call a function that executes a simple loop within I wait 1s. But the function is always interrupted and it returns -1.

     int i, ms;
     struct timespec	req;

     ms = 1000;

     for(i = 0; i < 5; i++) {
            printf("Counter %d\n\r", i);
            req.tv_sec = 1000 / 1000;
            req.tv_nsec = (1000 % 1000) * 1000000;
            nanosleep(&req, 0);   
     }

danielinux, 2017-03-06:
Nanosleep is not yet implemented (it's a stub in the libC that returns -1 and sets errno to ENOSYS).

danielinux, 2017-03-06:
This should be fixed now.

Please test with latest toolchain from https://archive.trexotech.com and the newest kernel.

nathanLoretan, 2017-03-06:
I've tested but the time defined in second into the struct timespec is not used.

danielinux, 2017-03-06:
My bad, there was a typo in the implementation. Should be fixed by the new toolchain (currently compiling, will be on the archive soon)

danielinux, 2017-03-06:
@nathanLoretan please try with latest toolchain
http://archive.trexotech.com:8081/

closed 2017-03-06 by danielinux

nathanLoretan, 2017-03-06:
It works, thank you.