insane-adding-machines/frosted #118: How using port uGFX

nathanLoretan, 2017-03-05:
Hello, I've heard that it is possible to use the graphic functions from ugfx library with the OS frosted. I want to know how can I link both?

danielinux, 2017-03-06:
This is currently a joint work in progress between insane-adding-machines and the ugfx.io community.
There is no port available yet, but we'll keep you informed when we have a working uGFX port compiling & working on frosted.

closed 2017-03-06 by danielinux

nathanLoretan, 2017-03-06:
I know. The port to bind uGFX and frosted is almost finished. I must implement the semaphore. But Joel Bodenmann told me that frosted has already implemented the framebuffer support to use graphic functions.

danielinux, 2017-03-06:
- compile frosted with framebuffer support
- open "/dev/fb0"
- set your RGB palette via     ioctl(fd_fb, IOCTL_FB_PUTCMAP,  ...);
- use write/lseek to fill the framebuffer