insane-adding-machines/frosted #120: Proper system clocks

brabo, 2017-03-08:
- [ ]  use jiffies and read that as CLOCK_MONOTONIC
- [ ] use rt offset and read/write that as CLOCK_REALTIME and for [s|g]ettimeofday()

danielinux, 2017-03-14:
Where was CLOCK_MONOTONIC supposed to be defined in kernel space? I had to add an explicit define in `sys.c` because I cannot find any reference to that symbol in kernel code. Missing common header?

brabo, 2017-03-14:
<time.h>

danielinux, 2017-03-14:
Time.h is nowhere to be found in kernel code. Is it a libc include? Libc include are ignored in kernel, as we compile with nostdinc.

brabo, 2017-03-14:
it is in newlib. that flag seems gone, compiles and works fine this way. i agree we should split it off to frosted-headers though and include that both ways.

danielinux, 2017-03-15:
The stdinc problem is not only related to this. This issue can be closed, but the problem persists until we fix #121 

Thanks

closed 2017-03-15 by danielinux